autocore-ai/MapToolbox #5: Export error when adding 'StopLine'

gwwang16, 2019-05-24:
Thank you so much for sharing this awesome maptool!
All functions work correctly and awesome here except for adding 'StopLine'.
There is an error during exporting vectormap when 'StopLine' element added. It says 'Object reference not set to an instance of an object', is there any step I missed? (cause there is no example of adding 'StopLine' in the demo video.
I'll appreciate it if you can give me any hints to solve it, thank you in advance.

Best,
Guangwei

The error info is the following
```
NullReferenceException: Object reference not set to an instance of an object
Packages.AutowareUnityTools.VectorMapTools.Export.VectorMapExtern.GetCsvData (Packages.AutowareUnityTools.VectorMapTools.StopLine stopLine, PointOctree`1[T] finalLanes, System.Collections.Generic.List`1[Packages.AutowareUnityTools.VectorMapTools.Export.CsvPoint]& csvPoints, System.Collections.Generic.List`1[Packages.AutowareUnityTools.VectorMapTools.Export.CsvLine]& csvLines, System.Collections.Generic.List`1[Packages.AutowareUnityTools.VectorMapTools.Export.CsvVector]& csvVectors, System.Collections.Generic.List`1[Packages.AutowareUnityTools.VectorMapTools.Export.CsvSignalLight]& csvSignalLights, System.Collections.Generic.List`1[Packages.AutowareUnityTools.VectorMapTools.Export.CsvStopLine]& csvStopLines) (at Library/PackageCache/com.autocore.map-toolbox@014173a8ec8f680781f0c764ba639bf583d5ab8b/VectorMapTools/Export/VectorMapExtern.cs:91)
Packages.AutowareUnityTools.VectorMapTools.Export.VectorMapExporter.ExportMaps (System.String folder) (at Library/PackageCache/com.autocore.map-toolbox@014173a8ec8f680781f0c764ba639bf583d5ab8b/VectorMapTools/Export/VectorMapExporter.cs:58)
Packages.AutowareUnityTools.VectorMapTools.Editor.EditorMenuExterns.ExportVectorMap () (at Library/PackageCache/com.autocore.map-toolbox@014173a8ec8f680781f0c764ba639bf583d5ab8b/VectorMapTools/Editor/EditorMenuExterns.cs:79)
```

[image]

MoeLang, 2019-05-24:
Oh sorry, stop line needs the end of Lane and Signal Light. So near the stop line should be two lanes connected at the stop line.  And you also need to drag a signal light to this stop line, At current version signal light should be the green ( pass ) signal position.
[image]

gwwang16, 2019-05-24:
Thank you very much, it works correctly now.

And, thank you again for providing this amazing tool 👍.

closed 2019-05-24 by gwwang16