autocore-ai/MapToolbox #6: cant add PCD files

likui01, 2019-06-04:
hi,
So i followed the video. added the dependencies and everything.
Selected the PCD folder but after that, i am unable to add file to a sample scene. I tried dragging the files but they are never shown in the  scene window. 
Its hard to tell from video, which shortcut is being used. 
Kindly if you could guide me.
thanks

MoeLang, 2019-06-04:
Console window may log something wrong such as file format unsupported. Make sure pcd in binary format and fields in "x y z | x y z intensity | x y z color" type.
If nothing in the console, your point cloud maybe far away from original point where camera focused，then double click on your point cloud object in hierarchy window, camera will move to this object.

likui01, 2019-06-15:
I have tried, still pointclod doesnt show

MoeLang, 2019-06-20:
Can you provide some image like #5 or pcd file in trouble ?

sharish33, 2019-06-20:
Hi, I have added the PCD data and files were uploaded successfully. But when i click on the file, I am not able to see the map on the screen. I tried loading the data several times but the same issue. First time when i loaded the data, i got the below error

Error : unknown field type fields x y z _ intensity ring _ unityengine.debug:logerror(object)

And this error didn't appear after the first load. Please can you help me out here in resolving the issue.

MoeLang, 2019-06-20:
> Hi, I have added the PCD data and files were uploaded successfully. But when i click on the file, I am not able to see the map on the screen. I tried loading the data several times but the same [...]
@sharish33 This tool only support binary pcd in field format "x y z" or "x y z intensity" and "x y z color", 
and you can try this code to get a pcd  supported.
`
pcl::PointCloud<pcl::PointXYZIR> input;
pcl::PointCloud<pcl::PointXYZI> output;
pcl::copyPointCloud(input, output);
`

sharish33, 2019-06-20:
>> @sharish33 This tool only support binary pcd in field format "x y z" or "x y z intensity" and "x y z color",
> > and you can try this code to get a pcd  supported.
> > `pcl::PointCloud<pcl::PointXYZIR> input; pcl::PointCloud<pcl::PointXYZI> output; pcl::copyPointCloud(input, output);`

@MoeLang , Thank you very much for your reply. I am little new to this Unity tool.
Yes My PCD is in Binary format and below is the header row for my PCD file:

**VERSION 0.7
FIELDS x y z _ intensity ring _
SIZE 4 4 4 1 4 2 1
TYPE F F F U F U U
COUNT 1 1 1 4 1 1 10
WIDTH 130819
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 130819
DATA binary**

And I believe my PCD file is in format "x y z intensity" by seeing the header file.Please correct me if I am wrong.

Also can you let me know that I need to add the above given code to the "manifest.JSON" file in package folder or any other file. Thank you in advance

MoeLang, 2019-06-20:
> Also can you let me know that I need to add the above given code to the "manifest.JSON" file in package folder or any other file. Thank you in advance

Haha, I mean write c++ code use pcl functions, load pcd file and save in another format. 
I will update a support for this if you're not in a hurry.
I mean “x y z intensity” field header like this:

# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z intensity
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 18796931
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 18796931
DATA binary

you can view [pcd read code](https://github.com/autocore-ai/MapToolbox/blob/master/PcdTools/PcdReader.cs#L58) which is lack of support for "x y z _ intensity ring _"

sharish33, 2019-06-20:
@MoeLang , Haha..Thank you for your help. For now i used a different tool to generate the vector map from PCD file. Looking forward to have support document. Keep doing the good work.

likui01, 2019-06-21:
@MoeLang  thank you for your help. The issue is solved. Now i have another question.
when i draw  lanes, the lanes are not connected when generate from maptool how to connect them so that it can be used by openplanner. The open planner fails to find the goal

MoeLang, 2019-06-21:
> @MoeLang thank you for your help. The issue is solved. Now i have another question.
> when i draw lanes, the lanes are not connected when generate from maptool how to connect them so that it can be used by openplanner. The open planner fails to find the goal

When dragging a start point of lane to end point of another lane, the start point will be attached to the end point in near range, and makes lanes connected automatically.

closed 2020-05-21 by MoeLang

kimth1757, 2022-01-13:
@likui01 hi, 
I'm having a same issue as you. I cannot import the .pcd data files.
How can you solve this issue?
Taehan.