autocore-ai/MapToolbox #9: Cannot import .PCD with x y z intensity format

ROS-lover, 2019-06-25:
Hello,

I first want to thank you for making this amazing tool available for free. :)
As mentioned, this format is supported 'x y z intensity' but I am not able to view the pcd file imported in unity. 

[image]

When I import it in unity 2019.1 using Autoware Toolkit, the points and heights pf the imported pcd are all 0 and indeed nothing is show once I drag it to Hierarchy tab.  

Here is the few first lines of the .pcd file
<>
"# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z intensity
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 83602
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 83602
DATA ascii
-118.03508 44.482094 -10.662604 100
-105.99339 45.155231 -10.440625 100
-208.71854 20.064911 -8.6673651 63
-109.67179 33.598198 -8.476162 21
-109.80906 33.615269 -8.4882936 21
-108.90988 33.954304 -6.9077754 53
-108.77178 33.96843 -6.8992209 17
0.72196484 82.771698 -6.8347192 14
-7.0545511 83.846542 -6.6522837 7
-6.8178158 84.107193 -6.759553 8
-3.307972 85.844109 -6.8757992 7
-3.7196136 88.503029 -6.9328365 8.5
0.14721179 88.590927 -6.7222276 9"
<>
Thanks in advance for the time!

MoeLang, 2019-06-25:
@ROS-lover  Ascii pcd not works, use `pcl_convert_pcd_ascii_binary <in.pcd> <out.pcd> 1` to save a binary format pcd, good luck!

ROS-lover, 2019-06-25:
Thanks @MoeLang ! the issue is solved! 
Cheers.

closed 2019-06-25 by ROS-lover