autocore-ai/MapToolbox #11: unity cannot resolving packages!

JokerJohn, 2019-07-08:
hello，when I  Create my new project in Unity 2018.2
 and add two lines to Packages/manifest.json dependencies，this error was thrown：
[image]

JokerJohn, 2019-07-08:
how can I solve it?

MoeLang, 2019-07-08:
@JokerJohn Load git url needs [Unity 2019.1](https://store.unity.com/download?ref=personal) or above version, it's an experiment.

JokerJohn, 2019-07-09:
ok，thank you

LinuxCup, 2019-09-03:
there was a problem when i fixed 'manifest.json' file.

like this:
Library\PackageCache\com.autocore.map-toolbox@4bc1255ed0587b7ad67b731890c90d53b66bff06\VectorMapTools\DoubleConnectBezier.cs(29,16): error CS0246: The type or namespace name 'PointOctree<>' could not be found (are you missing a using directive or an assembly reference?)Library\PackageCache\com.autocore.map-toolbox@4bc1255ed0587b7ad67b731890c90d53b66bff06\VectorMapTools\Lane.cs(27,16): error CS0246: The type or namespace name 'PointOctree<>' could not be found (are you missing a using directive or an assembly reference?)
Library\PackageCache\com.autocore.map-toolbox@4bc1255ed0587b7ad67b731890c90d53b66bff06\VectorMapTools\Lane.cs(28,25): error CS0246: The type or namespace name 'PointOctree<>' could not be found (are you missing a using directive or an assembly reference?)Library\PackageCache\com.autocore.map-toolbox@4bc1255ed0587b7ad67b731890c90d53b66bff06\VectorMapTools\Export\VectorMapCollection.cs(28,16): error CS0246: The type or namespace name 'PointOctree<>' could not be found (are you missing a using directive or an assembly reference?)
Library\PackageCache\com.autocore.map-toolbox@4bc1255ed0587b7ad67b731890c90d53b66bff06\VectorMapTools\Export\VectorMapExtern.cs(66,13): error CS0246: The type or namespace name 'PointOctree<>' could not be found (are you missing a using directive or an assembly reference?)

[image]

MoeLang, 2019-09-03:
[image]
@LinuxCup  Can you find this package?
[image]
manifest.json

LinuxCup, 2019-09-03:
thinks for your resply! my panel is different from yours, but i followed your instructions on the 'manifest.json' file.
[image]

[image]

LinuxCup, 2019-09-03:
i know this tool is used for autoware, is there any other may to make vecotr map? i am very interested in autonomous vehicle

MoeLang, 2019-09-03:
[image]
@LinuxCup 

Error message caused by this null reference, I guess assembly GUID changed during unity update, It works well in latest unity version (Unity 2019.1.14f1), next update I will use name instead of GUID.

There are many map format can use for autonomous HD map (vector map), such as OpenDRIVE/ Lanelet2(OpenStreetMap format). Autoware use vector map supported by AsiaTech, and they will changing format to lanelet2.

LinuxCup, 2019-09-04:
@MoeLang
Thanks for your explanation! 
 thank you again.

closed 2020-05-21 by MoeLang