autocore-ai/MapToolbox #12: 4 Way ruless

ROS-lover, 2019-07-08:
Hi,

I actually got to work with the tools you guys developed. It turned out to be very helpful. I can create nodes and lane for straight road but I am not sure how to create it for 4 ways.  I tried many ways but I could not get op_planner in autoware to work with the vector map I created at 3 or 4 ways cross sections. 

i.e. can a lane that ends at a 4 way cross section, be connected by 3 more lanes? going straight, left and right? otherwise without having to allocate another lane for each way, what are the possible way to hand this situation?

[image]

Thanks!

MoeLang, 2019-07-08:
@ROS-lover It works well with a lane connected with 4 lanes. Make sure 4 lane both auto connected to target lane end point in editor, you can feel the start point jump to target end point if they nearby enough.
[image]
[image]

ROS-lover, 2019-07-19:
Hi @MoeLang Thanks for the reply, I tried it and it works! But I do get a strange shape of lanes they look horizontally not leveled. (image bellow)
[image]
My .pcd which is binarized already, is simply a smooth surface but this is what I get... I am not sure if it effects the planner's performance.
Thanks again!

MoeLang, 2019-07-19:
@ROS-lover All lines stay close to road surface, you can view road surface by click pcd in hierarchy window, looks like a green grid mesh.
Open planner in autoware not care much about line height, but for other use make sure points map surface flat:)
[image]

ROS-lover, 2019-07-19:
@MoeLang  Thanks for the quick response, looks like my map's surface is not that flat. Do you happen to know how to smooth a surface of a pointcloud map we import? Thanks!

aicarmark, 2019-07-19:
@ROS-lover what tool you are using to build map?

ROS-lover, 2019-07-23:
@aicarmark  I am using NDT_mapping in Autoware.

MoeLang, 2019-07-29:
@ROS-lover ndt localizer relay on correct imu data and high quality lidar data, otherwise custom modifies are needed.

lihai88, 2020-04-28:
> @ROS-lover It works well with a lane connected with 4 lanes. Make sure 4 lane both auto connected to target lane end point in editor, you can feel the start point jump to target end point if they nearby enough.
> [image]
> [image]

please tell me how to set auto connect function,i can not find the setting.thanks.

lihai88, 2020-04-29:
It works well with a lane connected with 4 lanes. Make sure 4 lane both auto connected to target lane end point in editor, you can feel the start point jump to target end point if they nearby enough.

@MoeLang  please tell me how to set auto connect function,i can not find the setting.thanks.

closed 2020-05-21 by MoeLang