autocore-ai/MapToolbox #14: op_global_planner cannot read the vectormap lane data

ROS-lover, 2019-07-27:
Hello,

I have used this plugin on Unity and it worked for most of time time. However I cannot understand why these maps attached which are just lane data and differ from each other only slightly, won't work with op_global planner in autoware. map folder 19 and 23 work but not the rest. They only differ slightly from each other.

[drive-download-20190727T233322Z-001.zip](https://github.com/autocore-ai/MapToolbox/files/3438700/drive-download-20190727T233322Z-001.zip)

Terminal Error

process[op_global_planner-1]: started with pid [6166]
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
['roslaunch', 'op_global_planner', 'op_global_planner.launch', 'enableLaneChange:=False', 'enableReplan:=True', 'enableSmoothing:=True', 'enableRvizInput:=True', 'pathDensity:=1', 'velocitySource:=1']
pid=6539
sched policy=OTHER prio=0
Received Lanes704
Received Points1089
Received dtLanes739
Received Lines342
Received Nodes739
Map Version 2
>> Extracting Lanes ...
>> Fix Waypoints errors ...
>> Create Missing lane connections ...
terminate called after throwing an instance of 'std::out_of_range'
what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[op_global_planner-1] process has died [pid 6558, exit code -6, cmd /home/ocp/autoware/ros/install/op_global_planner/lib/op_global_planner/op_global_planner __name:=op_global_planner __log:=/home/ocp/.ros/log/76e8791a-afcc-11e9-b129-a4fc771a0d0d/op_global_planner-1.log].
log file: /home/ocp/.ros/log/76e8791a-afcc-11e9-b129-a4fc771a0d0d/op_global_planner-1*.log
... logging to /home/ocp/.ros/log/76e8791a-afcc-11e9-b129-a4fc771a0d0d/roslaunch-ocp-6539.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Thanks in advance for looking into it!

MoeLang, 2019-07-29:
@ROS-lover Can you provide the scene file (named by *.unity) with this problem ?

ROS-lover, 2019-07-29:
@MoeLang Thanks. Just updated the issue's attachment with the file requested placed in the vect_unity_test.zip file. Thank you!

MoeLang, 2019-07-31:
@ROS-lover Thank you! From the scene file I can't find obvious problems, it seems caused by precision lost during serialization and I'll dig into open planner.

ROS-lover, 2019-07-31:
@ROS-lover That's right, I also was quite surprised since I did not find any obvious reason in format of the file generated, is it possible it has to do with something related to open planner?

MoeLang, 2019-08-01:
@ROS-lover Yes, If open planner works wrong,  I should provide some fix. 
Currently all tests and fixes are aimed at Autoware ver 1.12

ROS-lover, 2019-08-01:
@MoeLang , thanks, I can switch to that version if the fix provided works.

MoeLang, 2019-08-01:
@ROS-lover Try this fix, it works well on my machines.
[image]

ROS-lover, 2019-08-01:
@MoeLang Thanks Moe, what you suggested works just fine with the input vectormap I have attached. I will try to see if I face a problem and can let you know. Thanks once again!

Fangwq, 2019-09-20:
@ROS-lover  Hi，I follow the your procedure and get the result. The op_global_planner process doesn't died and the vector map doesn't [mix up](https://github.com/autowarefoundation/autoware/issues/2307). However, the point map and vector map doesn't match. How to solve this problem?
[image]

ROS-lover, 2019-09-20:
Hi @Fangwq. The issue I posed was specifically for a vectormap i prepared regardless of an actual location, and without pointcloud map. So it's normal they pcd map and vector map dont match since they are not made on top of one another. About the open planner, I used an older version of autoware which needed a line of code to change suggested by Moe and looks like in the newer version it's not a problem. hope this addressed your concern.

closed 2020-05-21 by MoeLang