autocore-ai/MapToolbox #15: Error while Compiling (Unity 2019.1)

repa1030, 2019-08-21:
Hello,

I receive this error while adding Map Toolbox:

`Library\PackageCache\com.unity.burst@1.0.0\Editor\BurstAotCompiler.cs(633,32): error CS7036: There is no argument given that corresponds to the required formal parameter 'exceptionType' of 'BuildReport.AddMessage(LogType, string, string)'`

Thank you for your help

MoeLang, 2019-08-22:
@repa1030 Can you share me the scene file (named by *.unity) ? BrustAotCompiler error fixed in v0.0.9

repa1030, 2019-08-22:
Thank you for your answer.
Here is the Scene File and the Vector Maps Data:
[VectorMaps_and_SceneFile.zip](https://github.com/autocore-ai/MapToolbox/files/3528599/VectorMaps_and_SceneFile.zip)

MoeLang, 2019-08-22:
@repa1030 
I found '.' replaced by ','  but can not found the cause...
Right file was exported by v0.0.9 and loaded successfully.
[image]
[image]

repa1030, 2019-08-22:
Everything works fine now. Thank you 👍

closed 2019-08-22 by repa1030