autocore-ai/MapToolbox #16: Load PCD with error message: Mesh 'points': abnormal mesh bounds

CyanChiu, 2019-09-10:
Dear Sir,
I have a PCD file in FIELDS x y z intensity binary format as attached.
[road_pcd.zip](https://github.com/autocore-ai/MapToolbox/files/3594926/road_pcd.zip)
When I load this PCD file, I will get the error as below:
```
Mesh 'points': abnormal mesh bounds - most likely it has some invalid vertices (+/-inifinity or NANs) due to errors exporting.
Mesh bounds min=(-inf, -inf, -inf), max=(inf, inf, inf). Please make sure the mesh is exported without any errors.
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

```
But I don't know if there is any wrong format of my file, could you help me to check?

MoeLang, 2019-09-10:
@CyanChiu  This plugin support pcd format :
**FIELDS x y z intensity
SIZE 4 4 4 4
TYPE F F F F**

but this file in format :
**FIELDS x y z intensity
SIZE 4 4 4 1
TYPE F F F U**

CyanChiu, 2019-09-11:
Oh! I just checked the FIELDS and missed the data type of intensity. Thanks for your remind, I will transform this file and check again.

closed 2020-05-21 by MoeLang