autocore-ai/MapToolbox #19: Is there a plan to support wayarea?

BenjaminChi007, 2019-10-17:
Hi, Moe

Thanks for your work.
As mentioned on YouTube, 0.2 version is under cooking, I would like to ask that wayarea type is in coming support list?
Thanks.

Regards,
Benjamin

BenjaminChi007, 2019-10-21:
Hi Moe,

Thanks for your reply.
I am currently studying autoware.ai, and plan to use wayarea to do waypoint plan. The official tier4 vector map tool is terrible to use compared with your work.
Could you give some guide that how to add wayarea type based on your current work?
Further advise will be highly appreciated.
Thanks.

Regards,
Benjamin

BenjaminChi007, 2019-10-21:
@MoeLang 

Thanks for your reply.
I am currently studying autoware.ai, and plan to use wayarea to do waypoint plan. The official tier4 vector map tool is terrible to use compared with your work.
Could you give some guide that how to add wayarea type based on your current work?
Further advise will be highly appreciated.
Thanks.

Regards,
Benjamin

MoeLang, 2019-10-23:
@BenjaminChi007 Thanks for your intersting, add wayarea type maybe difficult if you new to Unity.
A new type inherited from Line, editor and export class are needed. I plan to update this repo next month and add wayarea.

BenjaminChi007, 2019-10-23:
@MoeLang ,

Thanks for your guide. 
I will try to learn to do the job myself first, and looking forward to your patch in the coming month :)
Thanks. 

Regards,
Benjamin

closed 2020-08-20 by MoeLang