autocore-ai/MapToolbox #20: Can't connect lane

dengnaitian, 2019-11-05:
Thank you so much for sharing this amazing tool.
When I draw the lane in this tool, I found that the number fo  BLID and FLID in lan.csv have a  misstake. The  lane1 can't connet to the lane2, for the reason that the FNID number is setted to 0 and the BLID is setted to 0. You can see the misstake  in the picture. I think the right situation is that FLID is  13 and the BLID is 12.
Thank you in advance.
[Vect_map.zip](https://github.com/autocore-ai/MapToolbox/files/3807986/Vect_map.zip)

[image]
[image]

MoeLang, 2019-11-06:
Thank you for reporting.
I develop and test this plugin with op_global_planner node, in this node, lanes can be fixed to connected lanes when start and end lane id filled with zero, even if when five or more lanes merge into a lane.
In old version, BLID FLID or BLID2 and so on are filled with lane id connected, but can only connect to less than five lanes.
Next step I will change format from current asiatech map to lanelet2,  current map format will be removed in the future.

ImCabbage, 2020-04-14:
@MoeLang 
Great to hear that lanelet2 will be supported in the future. 
When is that going to happen, is there a clear timeline?

closed 2020-05-21 by MoeLang