autocore-ai/MapToolbox #21: global path planning

trustmetao, 2019-12-16:
Given the end point with the vector drawn by Unity, can't we plan a suitable path based on multiple feasible paths?

[image]

MoeLang, 2019-12-17:
Hello @trustmetao , use op_global_planner and issue #12 may help you.

trustmetao, 2019-12-20:
Thanks for the quick response, I wanna know how to draw the traffic lights in your vector map tools. And In my vector map, only one main road is connected and the other roads are separated. A vector map like this cannot plan a path with the open planner, right?
[image]
[image]

MoeLang, 2019-12-20:
Yes, op planner needs connected lane, green bulb of traffic lights should be attached to target stop line script, reference to issue #5.

trustmetao, 2019-12-20:
How should I draw like this traffic lights in vector map tools? Besides, are crosswalk unnecessary in vector map?
[image]

trustmetao, 2019-12-23:
And the singaldata has no height imformation, just like pole, poledate. Are height imformation of trafficlights unnecessary in this tools?

MoeLang, 2019-12-23:
For current version of Autoware, pole is unnecessary in package trafficlight_recognizer. But I found a problem that trafficlight may not work in Autoware, and I will fix and update this tool this week.
方便的话欢迎添加个人微信“heli-heli”获取更详细帮助

closed 2020-08-07 by MoeLang