autocore-ai/MapToolbox #27: How to annotate crosswalk and traffic signs on pcd map?

HHF111, 2020-03-24:
Is there any way to annotate crosswalk and traffic signs on PCD map with the tool currently?

MoeLang, 2020-03-26:
@HHF111 Load Autoware demo vector map csv files as traffic signs example.
Crosswalk in coming!

HHF111, 2020-03-26:
Thanks for reply. If I have raw PCD map and I know where traffic sign and crosswalk are. How can I generate csv files for them? Thanks again

MoeLang, 2020-03-27:
> Thanks for reply. If I have raw PCD map and I know where traffic sign and crosswalk are. How can I generate csv files for them? Thanks again

[ReadMe](https://github.com/autocore-ai/MapToolbox#how-to-use)
* After load pcd, drag pcd to Hierarchy
* Create AutowareADASMap object, Inspector will provide functions like this:
[image]
Select target map element in hierarchy and select functions in inspector, modify they position in scene view.
Read [Unity Scene View Navigation](https://docs.unity3d.com/Manual/SceneViewNavigation.html) if you are not familiar with Unity.
Video tutorial available after more function added such as crosswalk and roadedge.