autocore-ai/MapToolbox #28: Unable to add git link in Package Manganer as Mentioned with current Master

gnykumar, 2020-03-25:
Hi,

I was using Vector Map builder from tierIV.
But when I found that this can be done using Unity, I came to use this
and tried with Unity 2019.3.4 and MapToolbox https://github.com/autocore-ai/MapToolbox/releases/tag/0.0.10-preview.3 Version. I was able to annotate vector map but when i tried yesterday.
below error is getting 
`System.DllNotFoundException: unity_debug
  at (wrapper managed-to-native) AutoCore.MapToolbox.PCL.NativePluginDebug.SetFunctionDebugLog(intptr)
  at AutoCore.MapToolbox.PCL.NativePluginDebug.RegisterDebugFunctions () [0x00028] in /home/yogeshkumargn/Desktop/Vector/test_2/Library/PackageCache/com.autocore.map-toolbox@6e688d3b43/Runtime/PCL/NativePluginDebug.cs:45 
  at AutoCore.MapToolbox.Editor.PCL.NativePluginEditorDebug..cctor () [0x00000] in /home/yogeshkumargn/Desktop/Vector/test_2/Library/PackageCache/com.autocore.map-toolbox@6e688d3b43/Editor/PCL/NativePluginEditorDebug.cs:28 
UnityEditor.EditorAssemblies:ProcessInitializeOnLoadAttributes(Type[]) (at /home/builduser/buildslave/unity/build/Editor/Mono/EditorAssemblies.cs:118)
`

First time when I tried the instruction to use is to add below lines in package.json
"com.autocore.map-toolbox": "https://github.com/autocore-ai/MapToolbox.git",
 "com.nition.unity-octree": "https://github.com/autocore-ai/UnityOctree.git#upm",

Now the instruction to use is to add below git link in Window -> Package Manager -> heavy_plus_sign -> Add package from git URL...
https://github.com/autocore-ai/MapToolbox.git.

Please Let me Know if I am missing anything.
Thanks for kind reply

MoeLang, 2020-03-26:
@gnykumar Tools updated and only support windows at the moment, change
```
"com.autocore.map-toolbox": "https://github.com/autocore-ai/MapToolbox.git", 
```
to
```
"com.autocore.map-toolbox": "https://github.com/autocore-ai/MapToolbox.git#0.0.10-preview.3", 
```
will clone old tag.

gnykumar, 2020-03-26:
Hi @MoeLang  ,

**Thanks** for your kind reply.

I will do this.

gnykumar, 2020-05-18:
In linux it worked with ur solution given.

But in windows10 not working
below error is coming and no option of autware maptoolbox 
show in menu bar
`System.DllNotFoundException: unity_debug
  at (wrapper managed-to-native) AutoCore.MapToolbox.PCL.NativePluginDebug.SetFunctionDebugLog(intptr)
  at AutoCore.MapToolbox.PCL.NativePluginDebug.RegisterDebugFunctions () [0x00028] in C:\Users\Flux Auto Robotics 1\Desktop\MapToolbox-master\MapToolbox-master\Runtime\PCL\NativePluginDebug.cs:45 
  at AutoCore.MapToolbox.Editor.PCL.NativePluginEditorDebug..cctor () [0x00000] in C:\Users\Flux Auto Robotics 1\Desktop\MapToolbox-master\MapToolbox-master\Editor\PCL\NativePluginEditorDebug.cs:28 
UnityEditor.EditorAssemblies:ProcessInitializeOnLoadAttributes(Type[])
`

Thanks for your kin

gnykumar, 2020-09-24:
@MoeLang  Thanks!

closed 2020-09-24 by gnykumar