autocore-ai/MapToolbox #30: load failed with return -1

Neilyooo, 2020-04-26:
can‘t load PCD file.it throw error like
`load failed with return -1
UnityEngine.Debug:LogError(Object)
AutoCore.MapToolbox.PCL.PointCloudReader:LoadPointCloudFile(String) (at E:/MapToolbox-0.1.1-preview.6/Runtime/PCL/PointCloudReader.cs:238)
AutoCore.MapToolbox.PCL.PointCloudReader:.ctor(String) (at E:/MapToolbox-0.1.1-preview.6/Runtime/PCL/PointCloudReader.cs:243)
AutoCore.MapToolbox.Editor.PCL.PointCloudImporter:OnImportAsset(AssetImportContext) (at E:/MapToolbox-0.1.1-preview.6/Editor/PCL/PointCloudImporter.cs:33)
UnityEditor.Experimental.AssetImporters.ScriptedImporter:GenerateAssetData(AssetImportContext)`

and my pcd file format like:
`FIELDS x y z intensity
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 1650105
HEIGHT 1`

MoeLang, 2020-04-26:
Project path or pcd path contains space character or non-English character ?

Neilyooo, 2020-04-26:
> Project path or pcd path contains space character or non-English character ?

Oh damn it!I just found my project path has non-English character.Thanks for your help