autocore-ai/MapToolbox #31: System.DllNotFoundException: unity_debug

jcheng1602, 2020-05-06:
System.DllNotFoundException: unity_debug
  at (wrapper managed-to-native) AutoCore.MapToolbox.PCL.NativePluginDebug.SetFunctionDebugLog(intptr)
  at AutoCore.MapToolbox.PCL.NativePluginDebug.RegisterDebugFunctions () [0x00028] in D:\demo2\Library\PackageCache\com.autocore.map-toolbox@e50e47803306d11e6e0186a70957503bc53c0b5a\Runtime\PCL\NativePluginDebug.cs:45 
  at AutoCore.MapToolbox.Editor.PCL.NativePluginEditorDebug..cctor () [0x00000] in D:\demo2\Library\PackageCache\com.autocore.map-toolbox@e50e47803306d11e6e0186a70957503bc53c0b5a\Editor\PCL\NativePluginEditorDebug.cs:28 
UnityEditor.EditorAssemblies:ProcessInitializeOnLoadAttributes(Type[])

gnykumar, 2020-05-18:
same issue :

`System.DllNotFoundException: unity_debug
  at (wrapper managed-to-native) AutoCore.MapToolbox.PCL.NativePluginDebug.SetFunctionDebugLog(intptr)
  at AutoCore.MapToolbox.PCL.NativePluginDebug.RegisterDebugFunctions () [0x00028] in C:\Users\Flux Auto Robotics 1\Desktop\MapToolbox-master\MapToolbox-master\Runtime\PCL\NativePluginDebug.cs:45 
  at AutoCore.MapToolbox.Editor.PCL.NativePluginEditorDebug..cctor () [0x00000] in C:\Users\Flux Auto Robotics 1\Desktop\MapToolbox-master\MapToolbox-master\Editor\PCL\NativePluginEditorDebug.cs:28 
UnityEditor.EditorAssemblies:ProcessInitializeOnLoadAttributes(Type[])
`

ivanchristina, 2020-05-18:
Also the same failure, when following the master readme step by step, does anyone have any clues?
Environment: Win10 64+ Unity 2019.3.13.  Unity 2020 also not works.

thesidjway, 2020-05-19:
Facing the same issue

MoeLang, 2020-05-21:
This occurred on first loaded, not effect normal usage and disappears after reopen your project.

gnykumar, 2020-05-21:
After adding MapToolBox, we do not get any option of autoware map toolbox in menu bar.
And we are getting the same error always.

Please help us by some guidelines of using this!

MoeLang, 2020-05-21:
@gnykumar Options in menu bar removed in current version.
Import PCD by right click in Project Panel and Import New Asset then select target pcd file
Create new map by right click in Hierarchy, Autoware -> AutowareADASMap

gnykumar, 2020-05-21:
Thanks @MoeLang  :-)

Andyinthesummer, 2020-09-18:
**Environment:** Ubuntu 16.04, unity 2019.4.10f1
It works well before importing pcd files, then I got the similar **issue**:
```
Asset import failed, "Assets/Scenes/bin_Laser-00147_-00846.pcd" > DllNotFoundException: unity_pcl
AutoCore.MapToolbox.PCL.PointCloudReader.LoadPointCloudFile (System.String file) (at Library/PackageCache/com.autocore.map-toolbox@00aba0b479/Runtime/PCL/PointCloudReader.cs:74)
AutoCore.MapToolbox.PCL.PointCloudReader..ctor (System.String file) (at Library/PackageCache/com.autocore.map-toolbox@00aba0b479/Runtime/PCL/PointCloudReader.cs:243)
AutoCore.MapToolbox.Editor.PCL.PointCloudImporter.OnImportAsset (UnityEditor.Experimental.AssetImporters.AssetImportContext ctx) (at Library/PackageCache/com.autocore.map-toolbox@00aba0b479/Editor/PCL/PointCloudImporter.cs:33)
UnityEditor.Experimental.AssetImporters.ScriptedImporter.GenerateAssetData (UnityEditor.Experimental.AssetImporters.AssetImportContext ctx) (at /home/bokken/buildslave/unity/build/Modules/AssetPipelineEditor/Public/ScriptedImporter.cs:22)
```

alperenbayraktar, 2020-11-23:
@MoeLang Still having the issue, this prevents the creation of mesh collider child object of the .pcd file. Only the "points" child is created and seen in the project panel. Thus the vector map won't snap into the mesh collider but always have y as 0. Can you fix the unity_debug dll?

alperenbayraktar, 2020-12-08:
Never mind, unity_debug issue is solved once you open the project again, for colliding I've opened a new issue #55

isFakeAccount, 2022-01-21:
I have the same issue on Pop OS 21.10. Restarting Unity did not fix the issue. I installed it on a Windows 10 PC and there were no issues.