autocore-ai/MapToolbox #32: How to align vector map with pcd points

Anderson0609, 2020-05-31:
When I configured the Maptools plugin, I found the following problems,
1. When adding a white line according to pcd, but changing the angle of view, it is found that the added white line and the original point of pcd are not at the same height;
[image]
[image]

2. How to add a curved line or lane? The current plugin seems to only add straight lines

How can I solve it? Thank you very much

MoeLang, 2020-06-01:
Height of lines needs drag by hands in current version.
To add a curved lane, click on a straight lane and click "Subdivision" ,move curve control points in Scene window, then click "Normal way" to apply subdivision.

Anderson0609, 2020-06-02:
Thank you very much for your reply. Are there any operation videos or tutorials about the latest version of the plugin? The operation method of the current version is very different from the operation method on YouTube(https://www.youtube.com/watch?v=WTRHPs8pN04), thank you.

MoeLang, 2020-06-03:
Sorry, new videos will update after few version.