autocore-ai/MapToolbox #35: DllNotFoundException: unity_pcl

dymons, 2020-07-07:
Hi! When I try to upload a PCD file, I get the following error
[image]

<details>
File format PCD

```
# .PCD v0.7 - Point Cloud Data file format
VERSION 0.7
FIELDS x y z intensity
SIZE 4 4 4 4
TYPE F F F F
COUNT 1 1 1 1
WIDTH 10000
HEIGHT 1
VIEWPOINT 0 0 0 1 0 0 0
POINTS 10000
DATA binary
```

</details>

I've read all the current issues, but couldn't find an answer to the solution of this error. Could you please help me to solve the problem?

closed 2020-07-21 by dymons

harderthan, 2020-07-27:
Hello, @dymons. Did you fix this problem?

ROS-lover, 2020-09-29:
Has anyone find a solution for this?