autocore-ai/MapToolbox #39: Roadmap about Maps and Map format

harderthan, 2020-07-29:
Hello, @MoeLang . 
I'm glad to find good util to edit maps and paths for Autoware.
I think that this tool increase productivity.

There was a meeting to discuss which map format is better or suitable to Autoware.
https://discourse.ros.org/t/autoware-maps-and-map-formats-working-group-meeting-minutes-24-july-2019/10059

The main point at the pose is that they are considering to use Lanelet2 format in the near future.
What is next plan for this tool? Do you have a roadmap about map format?

MoeLang, 2020-07-30:
Yes we have lanelet2 format editor and use lanelet2 maps for serial months. We are considering provide this editor after some fixes.

harderthan, 2020-08-03:
Could you share specific milestone or TODO-list for a couple of months?
I'll hope to contribute something if I can develop some features of your roadmap.

MoeLang, 2020-08-21:
Lanelet2 format supports are developing on branch [lanelet2](https://github.com/autocore-ai/MapToolbox/tree/lanelet2).
More tags and attributes will be added, welcome contributes and suggestions.

harderthan, 2020-09-23:
Thanks for the update, @MoeLang .
By the way, do you know LGSVL's Map Annotation? 
It can to create, edit, export of HD maps of existing 3D environments with Lanelet2.

Here is the [link](https://www.lgsvlsimulator.com/docs/map-annotation/)

What is different with this?