autocore-ai/MapToolbox #41: how to export .csv files

yuhang9803, 2020-07-30:
hi
when I finishing labeling the map with toolbox, how can i export the corresponding csv files ?
thank you

MoeLang, 2020-07-31:
Click "Autoware Vector Map” script last button "Save Vector Map into folder".

harderthan, 2020-08-04:
the red button on the "Inspector" tab

[image]

idontlikelongname, 2020-11-13:
> Click "Autoware Vector Map” script last button "Save Vector Map into folder".

thanks for you awesome map tools, I meet a issue when "Save Vector Map into folder".
When I was drawing map, to avoid crash, I was used to saving during drawing. 
but I find a issue, if save many times during saving, I can only record the elements (points or lines) during the lastest saving to now. but the elements before last saving are lost !!!!

hamlinzheng, 2021-03-31:
> thanks for you awesome map tools, I meet a issue when "Save Vector Map into folder".
> When I was drawing map, to avoid crash, I was used to saving during drawing.
> but I find a issue, if save many [...]

Maybe I meet the same problem, I can only save part of the data, How did you solve this problem?

closed 2021-07-14 by yuhang9803

idontlikelongname, 2021-08-24:
> Maybe I meet the same problem, I can only save part of the data, How did you solve this problem?

close the unity editor, reopen and save again
I think its bug in editor scripts