autocore-ai/MapToolbox #43: OpPlanner Can't Plan with Autocore Exported Vector Maps

yucedagonurcan, 2020-08-06:
Hello,

Thank you for this amazing tool. 

_Before this big release we were able to use MapToolBox perfectly for our planning needs._

But we are experiencing some issues about generating a global path when we use MapToolBox. 

You can see the gif:

[image]

Here I will give respective` lane.csv`:
[image]

`dtlane.csv:
`![dtlane](https://user-images.githubusercontent.com/16342629/89579420-6a395100-d83c-11ea-82c8-d66642f7ce91.png)

`point.csv`
[image]

What do you think is the issue here?

MoeLang, 2020-08-07:
The lane is too long for op planner,  make lanes subdivision like this:
[image]

yucedagonurcan, 2020-08-07:
Thank you, problem is resolved.

closed 2020-08-07 by yucedagonurcan