autocore-ai/MapToolbox #44: How to  create a loop with lane

yucedagonurcan, 2020-08-12:
Hi,

I wonder how can we create a loop with lanes? 

I am trying to stitch last lane and first lane of the GoSlicesLane

[image]

But it won't snap. 

Generally I want to know how can I connect two lanes(merge) into one lane? I know other issues related to this one but you can clearly see that I need to be able to snap them.

I tried to add another lane between last lane and first lane but no luck.

Is there any documentation/tutorial for new version? 

Have a greay day.

MoeLang, 2020-08-12:
GoSlicesLane can not connect itself by merge first and last lane.
For loop street like this, you'd better create 8 GoSlicesLane and connect them, 4 on straight and 4 on junction.
[image]

yucedagonurcan, 2020-08-12:
Thank you for your help, resolved the issue.

closed 2020-08-12 by yucedagonurcan