autocore-ai/MapToolbox #45: How to merge two lanes at the same node

Anderson0609, 2020-08-14:
Excuse me, I have a doubt about the plugin :
due to the vector map format in the csv file, how to merge two lanes at the same node in the following instance. 
[image]

MoeLang, 2020-08-14:
Only first and last node can used to merge lanes. In this instance you should drag first node of left two lanes to last node of right lane.

Anderson0609, 2020-08-14:
Do you mean the first node of left two lanes Node 1 and Node 2, the last node of right lane Node3， as  the following picture？
but how to merge the nodes？ thank you so much.
[image]

MoeLang, 2020-08-14:
Yes, just drag node1 and node2 to node3, they will auto connected in short distance.
[image]

Anderson0609, 2020-08-14:
Thank you so much for you demo operation. but at the last step , what operation did you do to the right lane? Just click the mouse or double click ？ any other Additional operations  is needed？ thank you for your kind reply.

Anderson0609, 2020-08-15:
I got it , thank you very much,

closed 2020-08-20 by MoeLang