autocore-ai/MapToolbox #47: Why the parameters such  B,L,H in point.csv are zero

doggyc, 2020-08-18:
the point should have the lat and lon information,why the generated point.csv have no such information in B,L,H

MoeLang, 2020-08-19:
B and L are not supported and H needs modify height of points (y in transform component).

doggyc, 2020-08-19:
> B and L are not supported and H needs modify height of points (y in transform component).
thanks,and I want to know where is the origin of the generated vector map,and how the Bx,Ly are caculated.

MoeLang, 2020-08-19:
PCD file should be placed in origin, then add vector map base on points cloud position.
So vector map origin is same as PCD origin.

doggyc, 2020-08-19:
> PCD file should be placed in origin, then add vector map base on points cloud position.
> So vector map origin is same as PCD origin.
so point cloud maps come with latitude and longitude information?

doggyc, 2020-08-19:
how can I set the origin in unity

MoeLang, 2020-08-19:
Just import PCD and drag PCD into Hierarchy, make sure Transform not moved.
[image]

doggyc, 2020-08-19:
> Just import PCD and drag PCD into Hierarchy, make sure Transform not moved.
> [image]

thanks a lot,do you have wechat ,I want  some more details.

MoeLang, 2020-08-19:
> thanks a lot,do you have wechat ,I want some more details.

wechat id: heli-heli

closed 2020-08-20 by MoeLang