autocore-ai/MapToolbox #49: Right way to create intersections and bidirectional lanes

yucedagonurcan, 2020-08-20:
Hello,

I want to create something like this:

[image]

But unfornutely open planner gives me no centerlines. I want to know what is the right way to construct this kind of intersections between lanes.

Also, can we create bidirectional lanes  in Autocore? Because if I draw two lanes with different directions close to each other, planning is not successfull.

Have a great day.

MoeLang, 2020-08-20:
This map is similar to your scenario, please load and have a look. 
[Map.zip](https://github.com/autocore-ai/MapToolbox/files/5105864/Map.zip)

Can you drag AutowareADASMap to Project window to make a prefab ( restore by right click on AutowareADASMap and "Unpack Prefab" ) then send me the prefab, thus I can find the problem.

yucedagonurcan, 2020-08-26:
Hello,

It is so frustrating that I drawed map from scratch for many times. Sometimes it works sometimes it doesn't like some kind of probabilistic model( bad one).

I don't see any centerlines wit op planner.

[AutowareADASMap.zip](https://github.com/autocore-ai/MapToolbox/files/5129666/AutowareADASMap.zip)

MoeLang, 2020-08-27:
I noticed the zigzag and make this lane straight .

[image]

[image]

[AutowareADASMap 1.zip](https://github.com/autocore-ai/MapToolbox/files/5133579/AutowareADASMap.1.zip)

yucedagonurcan, 2020-08-27:
Hello,

We did this on purpose, but I found the issue without changing that zigzag:

In the intersection there are two lane slices and their end points are close to each other. 

I modified the slices so that their end points far from each other and OP makes the global planning successfully.

closed 2021-06-03 by yucedagonurcan