autocore-ai/MapToolbox #50: No data in LinkID

gnykumar, 2020-09-23:
Hi,

I used MapToolbox successfully to annotate vector map and get that used in vector map loader
But to use in planner obejects like whitline, roadedge, stopline and other should have linkID filled
LinkID : IDs linking with Lane ID
And in the data which I got saved from this plugin using unity is not able to have linkIDs in vector map
csv files

Ex: 
whitline.csv from unity maptoolbox

**ID | LID | Width | Color | type | LinkID
-- | -- | -- | -- | -- | --
1 | 1 | 0.2 | W | 0 | 0
2 | 2 | 0.2 | W | 0 | 0
3 | 3 | 0.2 | W | 0 | 0
4 | 4 | 0.2 | W | 0 | 0**

whiteline.csv from autoware

**ID | LID | Width | Color | type | LinkID
-- | -- | -- | -- | -- | --
1 | 1 | 0.15 | W | 0 | 11327
2 | 2 | 0.15 | W | 0 | 2673
3 | 3 | 0.15 | W | 0 | 2671
4 | 4 | 0.15 | W | 0 | 2668**

Please let me know if am missing anything or
this is the actual output that maptoolbox currently
gives. And I want linkIDs indeed. Help to get this

Thanks in advance

[image]

MoeLang, 2020-09-28:
@gnykumar  LinkID not supported in this tool, and I cant find definition of LinkID. In Autoware demo map seems equal to nearest lane id ?

gnykumar, 2020-09-28:
Yes, LinkID are LaneID of the nearest lane, this utilization u can in vector map server and client package which is
mainly required for path planning to know the nearby assets to the traversing path(lane).
For the test case, you can Run autoware demo of loading 
1 Load autoware vector map
2 run vector map server 
3 run vector map client ( check and set current pose topic to gnss_pose)
4 run nmea_sentence to gnss pose pckg from gnss localizer
5 run their bag file
6 subscribe to vector map client topic in rviz 
and observe the objects fetching nearby the traversing lane.

Is it possible to get this LinkID in this tool?

MoeLang, 2020-09-28:
Some fixes are needed. For LinkID in whiteline.csv , nearest lane may either left side or right side one ?

gnykumar, 2020-09-28:
This I am also not sure,
I think it is based on drive mode,
Left side forward drive may near to left side whiteline
and 
Right side forward drive may near to right side whiteline

and LinkIDs are also needed in Stopline, Crosswalk, Signal data, roadsurfacemark.

gnykumar, 2020-09-30:
Hi @MoeLang ,

This is not relevant to this issue. But instead of increasing issues simply I am asking this :+1: 
Along with manually annotating vector map or lanlet2 map, Is it possible to have some options
to load some other format maps like osm, kml, kmz, xodr, lanlet2, and convert that to autoware vector map and lanelet2.

Ref: [Assure mapping tool](https://github.com/hatem-darweesh/assuremappingtools)

In the above ref, vector map format saving is not there, and I request that.

MoeLang, 2020-09-30:
Autoware vector map will be obsoleted, autoware new projects use lanelet2 instead of vector map.

You can try map annotation in [lgsvl simulator](https://github.com/lgsvl/simulator) witch support opendrive format.

gnykumar, 2020-10-01:
is annotation in lgsvl simulator is free to use?

gnykumar, 2020-10-05:
> Some fixes are needed. For LinkID in whiteline.csv , nearest lane may either left side or right side one ?

Is there any progress ?

commit 872417c5 (2020-10-10): Fixes #50
Add Nearest LaneID as LinkID
closed 2020-10-10 by MoeLang