autocore-ai/MapToolbox #51: unity_pcl not found when loading PCD files

ROS-lover, 2020-09-29:
Hey guys,

Found this issue on ubuntu 18 and 2020 and 2019.4 unity versions. As a follow up question: is there a way I could use the old version of this toolbox as it used to work for me? 

Thanks!

[image]

xqrkushnir, 2020-11-09:
Hi there,
You can pull a specific commit, branch, or tag as written here: https://docs.unity3d.com/Manual/upm-git.html (under `Specifying a package in a subfolder of the repository`).

I do confirm that I get the same issue for Ubuntu 18 and Unity 2019.4 LTS. I also can import PCD flawlessly in Windows 10 and Unity 2020.1.7f1.

alperenbayraktar, 2020-11-16:
Well, i fixed it by adding Windows build support to unity version i used.

xqrkushnir, 2020-11-16:
> Well, i fixed it by adding Windows build support to unity version i used.

Didn't do the trick for me. What version of Unity/Ubuntu do you have?

alperenbayraktar, 2020-11-17:
> 
> 
> > Well, i fixed it by adding Windows build support to unity version i used.
> 
> Didn't do the trick for me. What version of Unity/Ubuntu do you have?

Windows 10 Unity 2019.3.3f1

auronman, 2021-02-09:
HI, i have the same issue. DId you solve it in Ubuntu?

xqrkushnir, 2021-02-09:
> HI, i have the same issue. DId you solve it in Ubuntu?

Not really. I've imported a PCD map in Windows, then moved this project on Ubuntu