autocore-ai/MapToolbox #53: How to move and align the 'Signal' asset in v0.1.0

abhi5691, 2020-11-17:
Hi, 

Thanks for this awesome tool.
I am using the MapToolbox v0.1.0 with Unity 2019.4.14 on my Windows machine. I am able to insert an AutowareADASMap asset into my project hierarchy, and successfully add lanes, stoplines etc. Each of these assets adds ADASGoSlicesXX object for each lane/stopline instance.
But when I add a new Signal asset, there is a 'NewGameObject' added as a child to ADASMapSignal, and an ADASMapPole. I see a TL cone added at map origin, but I cannot move (neither rotate) this object by dragging. This makes it near impossible to precisely assign this asset to actual TLs in .pcd map.

Can you please help me with this or point me to a tutorial where you demonstrate adding a Traffic Light.

Thanks

alperenbayraktar, 2020-11-24:
You can also hover over the "Position X-Y-Z" label on the Transform component and drag them to left or right to adjust. But direction gizmos not showing is another problem. Can you see the direction gizmos with other kind of objects? Please try with Autoware objects and then just with default unity 3d-2d objects if possible.

alperenbayraktar, 2021-02-02:
it's a bit late but for those who have this problem, it's because setting Tools.current = Tool.None;
this mean setting this part of unity editor to no tool:
[image]
So, simply click on the move(2nd) tool to reveal handles.
Or create an editor class for those types and add this:
```
private void OnSceneGUI() //runs every frame when scene is opened
{
      Tools.current = Tool.Move;            
}

nhk035, 2021-08-18:
have a look at this https://github.com/autocore-ai/MapToolbox/issues/38#issuecomment-665379094