autocore-ai/MapToolbox #55: No Mesh Collider

alperenbayraktar, 2020-12-08:
@MoeLang  Old version used to have a collider child object under the imported pcd file's hierarchy view but for the moment it is not present. 
.pcd in project view: (only points as child)
[image]
.pcd in hierarchy view: (no children)
[image]

The vector map won't snap into the ground and will always have y = 0, this causes problems when you use tilted maps.

MoeLang, 2020-12-09:
Yes, I removed this function because this make our map cracked and uneven, we also fixed pcd height to zero.

alperenbayraktar, 2020-12-09:
Alright, but how are we supposed to draw the maps now? Because I can see that some of my pcd files have like 10m+ height difference from their start to end. And when you get a map file out, all the stuff are misplaced, like wayareas are 6m down under the ground. Should we rotate the map elements by hand? @MoeLang

MoeLang, 2020-12-09:
Config the endpoint height of line by hand, then subdivision. Dont rotate any map elements, just modity y axis value.

alperenbayraktar, 2020-12-09:
I see, it will be much time consuming unfortunately though. Thanks for fast answers!

alperenbayraktar, 2020-12-09:
@MoeLang Is there a way to bring back mesh collider snapping? I'm also editing the package and it could be awesome if I could use the first version's snapping combined with the new map editor scriptings

MoeLang, 2020-12-09:
This function in code [HeightMeshBuild](https://github.com/autocore-ai/MapToolbox/blob/0.0.7/PcdTools/HeightMeshBuild.cs) , I will add this feature back this weekend.

alperenbayraktar, 2020-12-09:
Okay, I'll test it asap, thanks!

ali-mehran-adastec, 2020-12-15:
> This function in c

THANKS: this would help a lot, though we can see the right and neat projection of vectors on image plane.  :)

alperenbayraktar, 2020-12-21:
> 
> 
> This function in code [HeightMeshBuild](https://github.com/autocore-ai/MapToolbox/blob/0.0.7/PcdTools/HeightMeshBuild.cs) , I will add this feature back this weekend.

Hey @MoeLang  did you add this to any branch? I can't find it.

alperenbayraktar, 2020-12-28:
Hey @moelang when are you going to add this?

moelang, 2020-12-28:
> Hey @moelang when are you going to add this?

Yes its under testing, some api changed and cannot used in old methods.

alperenbayraktar, 2020-12-28:
Alright thanks, have a nice day.

alperenbayraktar, 2021-01-11:
@moelang I tried the commit, now the first node gets the actual height but the second one doesn't upload its height when you move it and always has the same height as the first node. will you add a dynamic height setting for it or should I? 
[image]

moelang, 2021-01-12:
> @moelang I tried the commit, now the first node gets the actual height but the second one doesn't upload its height when you move it and always has the same height as the first node. will you add a dynamic height setting for it or should I?
> [image]

Subdevision added height from pcd in last commit

alperenbayraktar, 2021-01-12:
Now it works like a charm. Since setting height with subdivision (Adding ADASGoX), From and To go objects don't upload their heights after subdivision. If you add a lane and change the From position before subdivision, it looks like this, I've added some features myself but here's what I mean:
[image]

alperenbayraktar, 2021-01-12:
> 
> 
> @moelang Now it works like a charm. Since setting height with subdivision (Adding ADASGoX), From and To go objects don't upload their heights after subdivision. If you add a lane and change the From position before subdivision, it looks like this, I've added some features myself but here's what I mean:
> [image]

```
            float y = Utils.GetHeight(current.From);  //Snap the first one 
            current.From = new Vector3(current.From.x, y, current.From.z);
            foreach (var point in points)
            {                
                if (Vector3.Distance(current.From, point) > distance)
                {
                    Vector3 dest = current.From + (point - current.From).normalized * distance;
                    dest.y = Utils.GetHeight(dest);
                    current.To = dest;
                    var temp = current.AddAfter().GetComponent<LineSegment<T>>();
                    temp.DataCopy(current);
                    current = temp;
                    current.From = dest;
                }
            }
            y = Utils.GetHeight(current.To); //Snap the last one 
            current.To = new Vector3(current.To.x, y, current.To.z);
```

Added 4 lines to LineSegment.cs so now when subdivision is clicked, from and to heights are also set. @moelang

closed 2021-01-12 by alperenbayraktar