autocore-ai/MapToolbox #57: Lat Lon in Lanelet2

gnykumar, 2020-12-21:
Hi Team,

I am learning and exploring a lot from this tool in utilizing vector map and lanelet2 map. Thanks for the tool.

I seen autoware lanelet loader and visualizer at [autoware lanelet2](https://gitlab.com/autowarefoundation/autoware.ai/common/-/merge_requests/28) along with [sample](https://gitlab.com/autowarefoundation/autoware.ai/common/uploads/d35f15087676de4f26a583a9ff9eb41d/autoware_mapping_example.osm) lanelet2 map. There is lat lon values in map but from the map I am generating from MapToolbox doesn't have lat lon, I can see that there is no reference to provide that while or after annotating. To use the Lanelet2 lib for path planning, lat lon would be required, If someone helps me to get lat lon from MapToolbox based annotated lanelet2 map It will be great.

Thanks in Advance 

For your information, I found another tool called assure mapping tool using which we can  add projection information but map is getting changed a lot which is again not usable.

kuzen, 2021-02-18:
Hi!
You can refer to this pull request #63.

gnykumar, 2021-05-12:
HI @kuzen ,

I tried your Modification for JOSM from https://github.com/GDUT-IIDCC/MapToolbox

But I am getting below error for saving the Lanelet2 map
[image]

kuzen, 2021-05-12:
> HI @kuzen ,
> 
> I tried your Modification for JOSM from https://github.com/GDUT-IIDCC/MapToolbox
> 
> But I am getting the below error for saving the Lanelet2 map
> 
> [image]

Did you replace these files? And remember to restart Unity.
```
MapToolbox/Runtime/PCL/Geographic.dll
MapToolbox/Runtime/PCL/GeographicWarpper.dll
```

gnykumar, 2021-05-13:
Hi @kuzen , Those files are already there in your repo. I will pulled it newly and tried. But it is closing the application at Runtime.
[image]

kuzen, 2021-05-13:
Hi @gnykumar.
1. What coordinate system is the point in PCD (MRGS or UTM)? This modified version is suitable for pcd in MGRS coordinate. If you need to generate lanelet2 for pcd in UTM coordinates, you can modify line 113 in Node.cs.
2. Please confirm that your mgrs_code is correct.

gnykumar, 2021-05-13:
@kuzen  Ok Thanks, I Understood. And Currently, Map is not in MRGS or UTM. It is in local coordinates. But How to get PCD Map in MGRS Coordinate system. 
Long back we tried for UTM but face an issue in floating-point precision. Where in higher values, the PCD map was getting sparsed in points in the map.

Is there is any standard way that I can add MRGS coordinate to PCD?

We uses NDT, LIOSAM, LEGO LOAM for building PCD Map.

kuzen, 2021-05-13:
@gnykumar 
Just move the coordinate point. (x+x0, y+y0)
example (https://www.earthpoint.us/Convert.aspx):
point in UTM: 49Q 712345.12mE 2567890.34mN
point in MGRS: 49QGF12345126789034
point in PCD:  12345.12    67890.34
mgrs_code: 49QGF

gnykumar, 2021-05-13:
Thanks @kuzen

closed 2021-05-13 by gnykumar