autocore-ai/MapToolbox #59: Is there special rule when draw the lane in a corner

qiongwu1003, 2021-01-08:
Hi, all

I want to draw a vector map like below:
[image]

And I draw the lanes as below step:
[image]

Does the operation correct?

qiongwu1003, 2021-01-11:
Can anybody help me?
I drawed like above, but when i use the vector map in AutowareAI, the planning is not work.

moelang, 2021-01-11:
> Can anybody help me?
> I drawed like above, but when i use the vector map in AutowareAI, the planning is not work.

Please provide the csv files.

qiongwu1003, 2021-01-11:
Thanks for your reply.
This is the csv file.
[csv_file.zip](https://github.com/autocore-ai/MapToolbox/files/5794631/csv_file.zip)

And besides I use the master code.

moelang, 2021-01-11:
I tested your csv files in autoware.ai 1.14 and op_global_planner works well with outputs :

```
Received Lanes137
Received Lines6
Received Points154
Received dtLanes137
Received Nodes142
Map Version 2
>> Extracting Lanes ...
>> Fix Waypoints errors ...
>> Create Missing lane connections ...
>> Link lanes and waypoints with pointers ...
>> Extract Signal data ...
>> Extract Stop lines data ...
>> Link missing branches and waypoints...
>> Link StopLines and Traffic Lights ...
>> Map loaded from data with 5 lanes
```

qiongwu1003, 2021-01-11:
Thanks.
But I use the vector map refer to this [tutorial](https://www.lgsvlsimulator.com/docs/autoware-instructions/).
Is the vector map supports this mode?

moelang, 2021-01-11:
This branch only tested for op_global_planner and not updated after  autoware.1.14

qiongwu1003, 2021-01-11:
@moelang Thank you very much. No more question and close the issue.

closed 2021-01-11 by qiongwu1003