autocore-ai/MapToolbox #61: DllNotFoundException : unity_pcl

JinhyeonnJang, 2021-02-03:
I cant upload my binary pcd file on hierarchy,,,,

These are errors : 
Asset import failed, "Assets/0.3_prontier_binary.pcd" > DllNotFoundException: unity_pcl
AutoCore.MapToolbox.PCL.PointCloudReader.LoadPointCloudFile (System.String file) (at Library/PackageCache/com.autocore.map-toolbox@00aba0b479/Runtime/PCL/PointCloudReader.cs:74)
AutoCore.MapToolbox.PCL.PointCloudReader..ctor (System.String file) (at Library/PackageCache/com.autocore.map-toolbox@00aba0b479/Runtime/PCL/PointCloudReader.cs:243)
AutoCore.MapToolbox.Editor.PCL.PointCloudImporter.OnImportAsset (UnityEditor.Experimental.AssetImporters.AssetImportContext ctx) (at Library/PackageCache/com.autocore.map-toolbox@00aba0b479/Editor/PCL/PointCloudImporter.cs:33)
UnityEditor.Experimental.AssetImporters.ScriptedImporter.GenerateAssetData (UnityEditor.Experimental.AssetImporters.AssetImportContext ctx) (at /home/bokken/buildslave/unity/build/Modules/AssetPipelineEditor/Public/ScriptedImporter.cs:22)

Please help me,,,,,

moelang, 2021-02-03:
Windows 10 1803 or above are needed.

JinhyeonnJang, 2021-02-08:
THANKS A LOT!!!!!!!!!!!

closed 2021-02-08 by JinhyeonnJang