autocore-ai/MapToolbox #69: how to import pcd file to maptoolbox

yuhang9803, 2021-07-02:
hi , when i did as your document says to resolve the package , i found i could not import pcd file into the package, and the error will occur as the picture follows, in fact , when i want to import ,the unity could not find the pcd file in my computer
[image]
[image]

RI-DH, 2021-07-10:
I have the same problem. Have you solved it.

yuhang9803, 2021-07-10:
yeah，you need to check your version of unity   it's probably because of your version is not matched

RI-DH, 2021-07-11:
> yeah，you need to check your version of unity   it's probably because of your version is not matched 
> […](#)

Thank you, my problem is solved.

closed 2021-07-14 by yuhang9803