autocore-ai/MapToolbox #81: 错误	CS8377

MakerFace, 2022-06-15:
系统版本：Windows 11，21H2
Unity版本：2021.3.4f1c1、2019.4.39f1c1

严重性	代码	说明	项目	文件	行	禁止显示状态
错误	CS8377	类型“U”必须是不可为 null 值的类型，且包括任何嵌套级别的所有字段，才能用作泛型类型或方法“NativeList<T>”中的参数“T”	Unity.Jobs, Unity.Jobs.Player	D:\source\Unity Projects\lanelet2_map\Library\PackageCache\com.unity.jobs@0.8.0-preview.23\Unity.Jobs\IJobParallelForDefer.cs	77	活动

无法导入MapToolkit这个包。

q87956256, 2022-06-23:
> 系统版本：Windows 11，21H2 Unity版本：2021.3.4f1c1、2019.4.39f1c1
> 
> 严重性 代码 说明 项目 文件 行 禁止显示状态 错误 CS8377 类型“U”必须是不可为 null 值的类型，且包括任何嵌套级别的所有字段，才能用作泛型类型或方法“NativeList”中的参数“T” Unity.Jobs, Unity.Jobs.Player D:\source\Unity Projects\lanelet2_map\Library\PackageCache\com.unity.jobs@0.8.0-preview.23\Unity.Jobs\IJobParallelForDefer.cs 77 活动
> 
> 无法导入MapToolkit这个包。

你好，请问你解决了吗？ 我也是这个问题

MakerFace, 2022-06-23:
> 你好，请问你解决了吗？ 我也是这个问题

解决了，改成Windows 10，21H1，Unity版本 2019.4.0f1 Personal就可以导入了，然后从tags[0.2.0-preview.6](https://github.com/autocore-ai/MapToolbox/releases/tag/0.2.0-preview.6)下载到本地，从磁盘中加载就可以了，但是还是遇到一个NINNotFound的异常，应该不是什么问题，我就没管，可以正常编辑地图了。

closed 2022-06-23 by MakerFace

q87956256, 2022-06-23:
> 解决了，改成Windows 10，21H1，Unity版本 2019.4.0f1 [...]

好的，谢谢了 ，我试试