autocore-ai/MapToolbox #84: Ubuntu 20.04 import lanelet2_linux branch error

Kailthen, 2022-07-21:
Unity Version: 2019.4.40f1c1

Asset Packages/com.autocore.map-toolbox/Way.cs has no meta file, but it's in an immutable folder. The asset will be ignored.
[image]

kevinsu628, 2022-07-25:
Hi, I have the same issue. Did you resolve it?

Kailthen, 2022-07-28:
@kevinsu628 No, and I have tried a Unity 2021, got same error.

Kailthen, 2022-07-29:
New error message when "Load package form disk ...":
Library/PackageCache/com.unity.jobs@0.8.0-preview.23/Unity.Jobs/IJobParallelForDefer.cs(77,85): error CS8377: The type 'U' must be a non-nullable value type, along with all fields at any level of nesting, in order to use it as parameter 'T' in the generic type or method 'NativeList<T>'

Kailthen, 2022-07-29:
update  com.unity.jobs to 0.11.0-preview.5

closed 2022-07-29 by Kailthen

geun0196, 2023-07-25:
Hi Kailthen. I'm sorry about asking closed Issue.

Can I ask your Linux version??