autocore-ai/MapToolbox #91: Is Linux supported?

robodrome, 2023-02-17:
# Setup
- Ubuntu 22.04
- Unity3D Editor 2019.4.7f1 (same version as in screenshots in this repo readme)
- Template: 3D Core
- MapToolbox branch: main

# Description
I followed the [How to use](https://github.com/autocore-ai/MapToolbox#how-to-use) instructions. There are errors.

# Expected Behavior
No errors.

# Actual behavior
Errors:
```text
Class HashSetExtensions can not exist in multiple namespaces in the same file, even if one is excluded with preprocessor directives. Please move these to separate files if this is the case.
UnityEditor.Scripting.ScriptCompilers:GetClassAndNamespace(String, String, String&, String&) (at /home/bokken/buildslave/unity/build/Editor/Mono/Scripting/ScriptCompilers.cs:111)

Library/PackageCache/com.unity.collections@0.15.0-preview.21/Unity.Collections/NativeArrayExtensions.cs(50,63): error CS0426: The type name 'ReadOnly' does not exist in the type 'NativeArray<T>'

Library/PackageCache/com.unity.collections@0.15.0-preview.21/Unity.Collections/NativeArrayExtensions.cs(64,61): error CS0426: The type name 'ReadOnly' does not exist in the type 'NativeArray<T>'

Library/PackageCache/com.unity.collections@0.15.0-preview.21/Unity.Collections/NativeList.cs(716,31): error CS0426: The type name 'ReadOnly' does not exist in the type 'NativeArray<T>'
```

This is not my home directory by the way: `/home/bokken`

# Steps to reproduce
Follow instructions. I do see that Windows 10 is a requirement. Is that correct? A bit confusing is that I can also see Linux in the screenshots.

geun0196, 2023-07-25:
Hi, I have the issue too.

My environment is same below.
- Ubuntu 22.04
- Unity3D Editor 2021.3.28f1
- Template: 3D Core
- MapToolbox branch: lanelet2_linux

[image]

Did you solve issue?

robodrome, 2023-08-01:
No not yet. Unfortunately I currently do not have the time to look into it.